QUT-Motorsport-Admin/QUTMS_BMS_V2 #5: Fix voltage regulator having a maximum input voltage of 30V

liamkinne, 2020-01-16:
The pack can go above that. Rate the voltage regulator to 40V input and 5V output. You might need to use two regulators to drop the voltage in stages.

Whole 5V circuit can draw around 250mA, so it will need to be rated to that.

vivienmu, 2020-01-20:
Have found some options based on Geoffs recommendations. However, they are not rated to the necessary current. Matt said that another 3.3V output is needed for U5, so I would like to use two separate voltage regulators for the 5V and 3.3V. Not sure if placing them in parallel would drop the current through each one enough to make them suitable. The one I would like to use is this http://www.ti.com/lit/ds/symlink/lm5165-q1.pdf

vivienmu, 2020-01-20:
Also, there is no footprint or schematic model for this regulator available so I tried to make my own in Altium. I want someone to check the one I made please before I place it onto the schematic.