QUT-Motorsport-Admin/QUTMS_BMS_V2 #18: MCP2517 - interupt pins

SmugglerSMR, 2020-01-17:
Ideally, it will be useful to have INT, INT0 or INT1 pins connected to ATmega64M1 board. It is an unnecessary thing if not possible, but for some cases, better to have it.
Similar thing with CLKO/SOF one. But it is not clear for me what's its purpose.

vivienmu, 2020-01-20:
Where on the ATmega chip did you want the pin connected? There is currently only one unused pin on the chip, PIN3 (PC1), which according to the data sheet is an input pin, so I am not sure whether this is what you wanted. Is there any components needed to be placed within that connection?

SmugglerSMR, 2020-01-21:
If there is no PIN on Atmega, then so be it. Forget it. Interrupt pins are the least important part. You may close this issue.